marcelstoer/nodemcu-pyflasher #1: Waiting for the mac gui! :)

BLCKPSTV, 2017-01-12:


marcelstoer, 2017-01-12:
There _is_ a Mac GUI just not yet an `.app` bundle.

- git clone https://github.com/marcelstoer/nodemcu-pyflasher
- cd nodemcu-pyflasher
- python nodemcu-pyflasher.py

BLCKPSTV, 2017-01-12:
I managed to do this but got following error. 
File "nodemcu-pyflasher.py", line 3, in <module>
    import Main
  File "/Users/quintendewilde/nodemcu-pyflasher/Main.py", line 3, in <module>
    import wx
ImportError: No module named wx

marcelstoer, 2017-01-12:
That is to be expected because running it directly requires [wxPython to be installed in your Python environment](https://wiki.wxpython.org/How%20to%20install%20wxPython#Mac_OS_X).

BLCKPSTV, 2017-01-12:
Ok, so I did 

` sudo pip install --upgrade --trusted-host wxpython.org --pre -f http://wxpython.org/Phoenix/snapshot-builds/ wxPython_Phoenix `

marcelstoer, 2017-01-12:
I linked to the stable release install for a reason. I didn't program against wxPython phoenix.

```
~/Data/Python/NodeMCU-PyFlasher (master) > python
Python 2.7.12 (default, Oct 11 2016, 05:24:00) 
[GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.38)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import wx
>>> wx.version()
'3.0.2.0 osx-cocoa (classic)'
```

Oh, and of course you'd still [need esptool installed](https://github.com/espressif/esptool#easy-installation).

I really hope they're going to fix https://github.com/pyinstaller/pyinstaller/issues/2355 soon so this hassle will go away for Mac users.

BLCKPSTV, 2017-01-12:
I have esptool installed, I think I followed your stable link (but that might not be your point)
I really hope so to, I've tried so many ways to get micropython etc on my WemosD1miniPro

marcelstoer, 2017-01-12:
If you follow my initial wxPython link you'll end up on https://www.wxpython.org/download.php where they offer wxPython3.0-osx-cocoa-py2.7.

BLCKPSTV, 2017-01-12:
Ow it seems I forgot to tell another error in my steps, when downloading the pkg for cocoa or carbon, OSX Sierra tells me the package is damaged

BLCKPSTV, 2017-01-12:
The installer could not be installed because it couldn't find the software is the result of installing the pkg, do you have the pkg that works?

marcelstoer, 2017-01-12:
I downloaded it from there as well, sorry can't help with that 😞

BLCKPSTV, 2017-01-15:
I've managed to install it with a program called Pacifist, (I see it when I look at the packages via pip.freeze) I get 
```
wxPython==3.0.2.0
wxPython-common==3.0.2.0
wxPython-Phoenix==3.0.3.dev2749+f803d20
```

So clearly some versions got installed, except for the one from the link you gave me. 
wondering why it still doesn't work though

blackbyte-pl, 2017-01-23:
I had the same problem as @BLCKPSTV . I managed to install wxPython from homebrew with simply
`brew install wxpython`

but it didn't solve the problem - I couldn't run app from `dist` folder. I was able to run it from main dir though:
`python Main.py`

work like a charm

marcelstoer, 2017-01-23:
> I couldn't run app from `dist` folder.

That's expected and I never claimed you could. It's right here: https://github.com/marcelstoer/nodemcu-pyflasher#status. If you can help "promote" https://github.com/pyinstaller/pyinstaller/issues/2355 by leaving comments there it'd probably be beneficial for quite a few people.

> please consider a section in readme when explaining what to install and do before running the app and how to run it.

Sure, can do that. However, I personally don't see why anybody on Mac would try to use this app (as long as there's no app bundle, see above) rather than running `esptool.py` directly from the console. Do you feel having a GUI justifies the extra effort of installing Python bundles?

ozarchie, 2017-01-25:
On Mac OSX, `brew install wxpython` worked.
At the end of the install, the following message was included:  

  Python modules have been installed and Homebrew's site-packages is not
in your Python sys.path, so you will not be able to import the modules
this formula installed. If you plan to develop with these modules,
please run:
`
  mkdir -p /Users/xxxx/Library/Python/2.7/lib/python/site-packages  
  echo 'import site; site.addsitedir("/usr/local/lib/python2.7/site-packages")' >> 
 /Users/xxxx/Library/Python/2.7/lib/python/site-packages/homebrew.pth  
`

After that, `./nodemcu-pyflasher.py` worked.

solancer, 2017-03-12:
Left a comment on pyinstaller/pyinstaller#2355 hope this gets resolved soon..

siva1024, 2017-08-17:
I Changed some of the code in the main.py for supporting the new wxPython 4.0 (Phoenix) the codes in the link below. I also made an application file for macOS direct installation. I just checked it and it is working fine. I am going to post the code I modified within the same link if anyone wants they can have a look
https://www.dropbox.com/sh/a45xd2mco7trmjp/AACI93iTN120uMvjslWz-EwMa?dl=0

marcelstoer, 2017-08-18:
@siva1024 
1. Cool!
2. The about dialog doesn't work, at least not in the compiled macOS binary.
3. Can you please add the script you used for building macOS binary to Dropbox as well.

siva1024, 2017-08-18:
I just uploaded the script that i used for compiling in the same dropbox folder

kskenyon, 2017-09-28:
There is a 2.1 version now and I can't figure out what you changed in Main.py.  I tried compiling it with the version on GitHub using your setup.py but it doesn't work.  Something about images is messed up.  

Would you mind doing a pull and posting a compiled binary for those of us that are struggling to get a version of nodemcu-pyflasher.app with the dout option?

Thanks

marcelstoer, 2017-09-29:
The "you" in your previous comment seems to be directed at @siva1024 rather than at me. FWIW to see _my_ changes since 2.0 you can go to the release and look for "n commits to master since this release". This will bring you to https://github.com/marcelstoer/nodemcu-pyflasher/compare/v2.0...master

kskenyon, 2017-09-29:
Yes, "you" meant @siva1024.  Sorry for the confusion.

I did take @siva1024's edits to Main.py for wxPython 4.0 and applied them to marcelstoer's update in GitHub.  I got it to run after a couple of tweaks and used pyinstaller -w to make a macOS .app file and it works fine.  The desktop icon is a bit messed up and so are the images in the About box but I didn't fix that.  The Main.py changes were easier to find than I thought at first. The dout addition is good.  On to the next challenge.  

[image]

marcelstoer, 2017-09-29:
Looking forward to wxPython 4 GA as I've given up any hope that pyinstaller/pyinstaller#2355 will ever be fixed. Thanks for confirming that wxPython 4 is not affected by that bug.

marcelstoer, 2018-02-10:
Fixed with the latest [3.0 release](https://github.com/marcelstoer/nodemcu-pyflasher/releases/tag/v3.0). Enjoy!

closed 2018-02-10 by marcelstoer

kskenyon, 2018-02-10:
private$ python nodemcu-pyflasher.py 
objc[20812]: Class FIFinderSyncExtensionHost is implemented in both /System/Library/PrivateFrameworks/FinderKit.framework/Versions/A/FinderKit (0x7fffa2b69cb0) and /System/Library/PrivateFrameworks/FileProvider.framework/OverrideBundles/FinderSyncCollaborationFileProviderOverride.bundle/Contents/MacOS/FinderSyncCollaborationFileProviderOverride (0x115ca3cd8). One of the two will be used. Which one is undefined.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/Users/private/nodemcu-pyflasher/Main.py", line 76, in run
    esp = ESPLoader.detect_chip(self._config.port, initial_baud)
  File "/Users/private/nodemcu-pyflasher/esptool.py", line 216, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "/Users/private/nodemcu-pyflasher/esptool.py", line 195, in __init__
    self._set_port_baudrate(baud)
  File "/Users/private/nodemcu-pyflasher/esptool.py", line 200, in _set_port_baudrate
    self._port.baudrate = baud
AttributeError: 'unicode' object has no attribute 'baudrate'
This happens when I click on the Flash Button in 3.0.  My version of 2.5 is working fine.

Kevin

marcelstoer, 2018-02-10:
If you run it through your own Python interpreter using `python nodemcu-pyflasher.py` then you need to make sure you install esptool.py v2.2.1 manually! The `esptool.py` is only used during the packaging phase.