marcelstoer/nodemcu-pyflasher #12: Fatal Error - Failed to open script nodemenu-pyflasher

GeoffState, 2017-03-09:
Error occured so I deleted the dist\NodeMCU-PyFlasher.exe and tried again as below. Tried running exe as admin, no change.

G:\Downloads\nodemcu-pyflasher-master>build

G:\Downloads\nodemcu-pyflasher-master>pyinstaller --log-level=DEBUG             --noconfirm             build-on-win.spec
182 INFO: PyInstaller: 3.2.1
182 INFO: Python: 3.5.2
183 INFO: Platform: Windows-10-10.0.14393-SP0
184 DEBUG: Testing for UPX ...
186 INFO: UPX is not available.
188 DEBUG: script: nodemcu-pyflasher.py
188 INFO: Extending PYTHONPATH with paths
['G:\\Downloads\\nodemcu-pyflasher-master']
189 INFO: checking Analysis
215 INFO: Appending 'datas' from .spec
221 INFO: checking PYZ
248 INFO: checking PKG
250 INFO: Building because toc changed
251 INFO: Building PKG (CArchive) out00-PKG.pkg
392 DEBUG: Compiling c:\users\geoff\anaconda3\lib\site-packages\PyInstaller\loader\pyiboot01_bootstrap.py
392 DEBUG: Compiling c:\users\geoff\anaconda3\lib\site-packages\PyInstaller\loader\pyiboot01_bootstrap.py
395 DEBUG: Compiling G:\Downloads\nodemcu-pyflasher-master\nodemcu-pyflasher.py
396 DEBUG: Compiling G:\Downloads\nodemcu-pyflasher-master\nodemcu-pyflasher.py
2315 INFO: Building PKG (CArchive) out00-PKG.pkg completed successfully.
2326 INFO: Bootloader c:\users\geoff\anaconda3\lib\site-packages\PyInstaller\bootloader\Windows-64bit\runw.exe
2327 INFO: checking EXE
2329 INFO: Rebuilding out00-EXE.toc because NodeMCU-PyFlasher.exe missing
2329 INFO: Building EXE from out00-EXE.toc
2358 INFO: SRCPATH [('images\\icon-256.ico', None)]
2358 INFO: Updating icons from ['images\\icon-256.ico'] to C:\Users\Geoff\AppData\Local\Temp\tmpevfm734m
2361 INFO: Writing RT_GROUP_ICON 0 resource with 20 bytes
2361 INFO: Writing RT_ICON 1 resource with 26144 bytes
2390 INFO: Appending archive to EXE G:\Downloads\nodemcu-pyflasher-master\dist\NodeMCU-PyFlasher.exe
2573 INFO: Building EXE from out00-EXE.toc completed successfully.

marcelstoer, 2017-03-09:
I don't understand what's going on here. 

`G:\Downloads\nodemcu-pyflasher-master` is this cloned repository, right?
Running `build` (actually `build.bat`) in that directory should trigger PyInstaller. I don't see any output in your description, was there?
So, you eventually executed the statement from `build.bat` manually. That seems to have worked.

And then?

Can you successfully run the binary offered for download at https://github.com/marcelstoer/nodemcu-pyflasher/releases?

GeoffState, 2017-03-09:
nodemcu-pyflasher-master.zip 16/02/17 04:01 1,452 KB

https://github-cloud.s3.amazonaws.com/releases/77565174/f5d96452-d787-11e6-8

 

Ran build.bat from explorer first time

Ran build.bat from command line second time and copied output

 

Both times NodeMCU-PyFlasher.exe produced this message

cid:image001.png@01D29906.E02ACAC0

 

 

Thanks for binary download. That works fine.

I wish I could have found that insignificant “2 releases” earlier. I didn’t know I was looking for that magic word.

I have really come to hate Github recently. I thought I had to Clone or Download.

Thanks for educating me.

 

BTW. I just found a another effort I tried on 24 Feb. I must have just given up that time but left zip extract and built files in place on my C: drive. That NodeMCU-PyFlasher.exe gives the same message.

 

I can run further testing on download if you instruct.

 

Best Regards
Geoff Jenkins

marcelstoer, 2017-03-29:
If you want to build it yourself you need to 

- install Python
- install pyserial
- install wxpython
- clone this repository
- run `build.bat` in the cloned repo folder

closed 2017-03-29 by marcelstoer