marcelstoer/nodemcu-pyflasher #17: can it be done via pip installation ?

xiaolaba, 2017-06-16:
hi, there is able to complete esptool & wxpython by uses python and the pip.
how about your great piece of software, this asking is because we are new to python and have no idea how to uses your tools to build the GUI under windows.
thanks.

marcelstoer, 2017-06-16:
I have my doubts that I really understand your request.

No, pip installation is not supported. If you want to build NodeMCU-PyFlasher yourself (why not download the .exe that is available?) you have to
- install Python
- install wxpython
- install pyserial (I think)
- clone this repository and run `build.bat`

xiaolaba, 2017-06-17:
thanks for the response. yes, the exe downloaded is ok here. however, try a local build is able to be complete, but the exe had been failed on the fire, so we doubts that about build process variation and problem on this python build learning.

the specific problem is like this encountered,
https://xiaolaba.files.wordpress.com/2017/06/14mb-but-no-executable-gui.jpg

closed 2017-07-15 by marcelstoer