marcelstoer/nodemcu-pyflasher #19: I compiled a version for my mac

detectingdragon, 2017-07-24:
My mac is running Mac OS X 10.9.5, if i share the application, can someone else use it? (On their own computer)

marcelstoer, 2017-07-24:
How did you compile it? Feel free to attach/upload the binary to this "issue".

detectingdragon, 2017-07-24:
I did the command ./build.sh

detectingdragon, 2017-07-24:
Im uploading the result now.

detectingdragon, 2017-07-24:
But it wont let me launch it: 
Traceback (most recent call last):
  File "nodemcu-pyflasher.py", line 3, in <module>
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
    exec(bytecode, module.__dict__)
  File "Main.py", line 3, in <module>
    import aiohttp
ImportError: No module named 'wx'
Failed to execute script nodemcu-pyflasher
logout

detectingdragon, 2017-07-24:
Here is the link: https://mega.nz/#F!bAQzXJYA!S04pmM2RN76Nmi7ygw3O8g

marcelstoer, 2017-07-24:
If you want to build a binary you need to have wxPython and esptool (which brings pyserial) installed. See https://github.com/marcelstoer/nodemcu-pyflasher/issues/1#issuecomment-272253609 for details.
Also consider https://github.com/marcelstoer/nodemcu-pyflasher#status -> doesn't seem possible to have a working Mac binary.

closed 2017-07-24 by marcelstoer