marcelstoer/nodemcu-pyflasher #20: Support Flash Mode Dout

arendst, 2017-08-09:
I would like to advertize your tool for my [Tasmota](https://github.com/arendst/Sonoff-Tasmota) project as it seems to be a single container executable.

To get rid of all kinds of issues regarding different hardware (ESP8266 vs ESP8285) I decided to only support Flash Mode Dout as it works fine on both chips and all Tasmota supported hardware.

Would you consider implementing the Dout option in your Gui?

marcelstoer, 2017-08-09:
Sure, why not. For consistency reasons I'd have to support all four flash modes  rather than just three. Yet, I actually see no need for QOUT.

marcelstoer, 2017-08-16:
@arendst what's your take on the question I raised in the above comment?

arendst, 2017-08-16:
I'm not interested in QOUT. I'm only interested DOUT as that is the only way to go in my future supported hardware with Tasmota as the manufacturers have massively chosen for esp8285 which only supports DOUT.

marcelstoer, 2017-08-16:
I know but the big picture is bigger than your specific use case.

arendst, 2017-08-16:
Well I asked with regards to my use case. If it's too much of a problem alright with me. Will look for another tool or clone yours and tune to my likings.

Thnx.

marcelstoer, 2017-08-16:
It ain't "too much of a problem" and there's no reason to get annoyed. Every change offers the opportunity to (re-)evaluate an existing feature slightly beyond the one specific problem that triggered the change in first place.

arendst, 2017-08-16:
Agree.

ThomDietrich, 2017-08-17:
@marcelstoer good thinking on your side and I'd also not see the need. From what I read, @arendst is of course not against QOUT, it just doesn't matter to him. Maybe he got the question wrong. 
Anyhow, I'd tend to say that you could just add all four methods as option and be done with it. Either choice is fine and depends on your willingness to implement and test everything. Thanks for your effort.

marcelstoer, 2017-09-15:
Looking forward to your feedback, here's a binary for testing:
[NodeMCU-PyFlasher-dout-support.exe.zip](https://github.com/marcelstoer/nodemcu-pyflasher/files/1307572/NodeMCU-PyFlasher-dout-support.exe.zip)

closed 2017-09-15 by marcelstoer

arendst, 2017-09-15:
Thnx. Will look in to it soon and let you know.

marcelstoer, 2017-09-16:
Thanks for the feedback.

> Would it be possible to also provide the possibility to select the desired flash size in the GUI.

It would but it's highly unlikely that I'll implement that. I actually see value in providing as few knobs to turn as possible, see https://github.com/marcelstoer/nodemcu-pyflasher#why-this-project-exists to understand my motivation. Ideally it should just work w/o the user having to worry about the gory details.

Related to #11.

ThomDietrich, 2017-09-16:
"as few knobs as possible" -> How about abstract presets then? Currently the UI asks for Baud rate and Flash mode but the average user will not even be able to decide that. Apart from these "Custom" settings, there could be a first selection element offering presets for famous devices or projects like "NodeMCU devkit 1.0", "WeMos D1 mini", or "Sonoff-Tasmota".
I feel like this would be a huge step in terms of user experience and the implementation effort seems reasonable... wdyt?

marcelstoer, 2017-09-17:
I've been thinking about something similar in the past but didn't spend too much time developing the idea in my head. This time I did and I think it could turn out to be quite cool. Please open a separate issue so we can continue the discussion there.

commit 0116a91e (2017-09-15): Add flash mode 'dout'

Fixes #20
commit db84ce52 (2017-09-26): Add flash mode 'dout'

Fixes #20