marcelstoer/nodemcu-pyflasher #21: Are there directions?

jpan127, 2017-08-12:
Hi, just wondering are there directions for this application?

I am on windows 10, and used Git Bash to run the command `./build.sh` which I'm guessing is what we are supposed to do.

However I got an error and I don't recognize it:

```$ ./build.sh
105 INFO: PyInstaller: 3.2.1
105 INFO: Python: 3.5.3
105 INFO: Platform: Windows-10-10.0.15063-SP0
105 DEBUG: Testing for UPX ...
106 INFO: UPX is not available.
107 DEBUG: script: nodemcu-pyflasher.py
107 INFO: Extending PYTHONPATH with paths
['C:\\Users\\JP\\Desktop\\nodemcu-pyflasher-master']
107 INFO: checking Analysis
116 INFO: Appending 'datas' from .spec
118 INFO: checking PYZ
132 INFO: checking PKG
133 INFO: Building because toc changed
133 INFO: Building PKG (CArchive) out00-PKG.pkg
175 DEBUG: Compiling c:\python35\lib\site-packages\PyInstaller\loader\pyiboot01_bootstrap.py
175 DEBUG: Compiling c:\python35\lib\site-packages\PyInstaller\loader\pyiboot01_bootstrap.py
176 DEBUG: Compiling C:\Users\JP\Desktop\nodemcu-pyflasher-master\nodemcu-pyflasher.py
176 DEBUG: Compiling C:\Users\JP\Desktop\nodemcu-pyflasher-master\nodemcu-pyflasher.py
1558 INFO: Building PKG (CArchive) out00-PKG.pkg completed successfully.
1564 INFO: Bootloader c:\python35\lib\site-packages\PyInstaller\bootloader\Windows-64bit\runw.exe
1564 INFO: checking EXE
1564 INFO: Building EXE because out00-EXE.toc is non existent
1564 INFO: Building EXE from out00-EXE.toc
1586 INFO: SRCPATH [('images/icon-256.icns', None)]
1586 INFO: Updating icons from images/icon-256.icns to C:\Users\JP\AppData\Local\Temp\tmpl6b0n8y1
Traceback (most recent call last):
  File "c:\python35\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\python35\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python35\Scripts\pyinstaller.exe\__main__.py", line 9, in <module>
  File "c:\python35\lib\site-packages\PyInstaller\__main__.py", line 90, in run
    run_build(pyi_config, spec_file, **vars(args))
  File "c:\python35\lib\site-packages\PyInstaller\__main__.py", line 46, in run_build
    PyInstaller.building.build_main.main(pyi_config, spec_file, **kwargs)
  File "c:\python35\lib\site-packages\PyInstaller\building\build_main.py", line 788, in main
    build(specfile, kw.get('distpath'), kw.get('workpath'), kw.get('clean_build'))
  File "c:\python35\lib\site-packages\PyInstaller\building\build_main.py", line 734, in build
    exec(text, spec_namespace)
  File "<string>", line 27, in <module>
  File "c:\python35\lib\site-packages\PyInstaller\building\api.py", line 411, in __init__
    self.__postinit__()
  File "c:\python35\lib\site-packages\PyInstaller\building\datastruct.py", line 161, in __postinit_
    self.assemble()
  File "c:\python35\lib\site-packages\PyInstaller\building\api.py", line 498, in assemble
    icon.CopyIcons(tmpnm, self.icon)
  File "c:\python35\lib\site-packages\PyInstaller\utils\win32\icon.py", line 172, in CopyIcons
    hsrc = win32api.LoadLibraryEx(srcpath, 0, LOAD_LIBRARY_AS_DATAFILE)
pywintypes.error: (193, 'LoadLibraryEx', '%1 is not a valid Win32 application.')
```

Thanks!

marcelstoer, 2017-08-12:
Hmmm, what makes you think you need to build the application yourself? There are runnable binaries at https://github.com/marcelstoer/nodemcu-pyflasher/releases for Windows.

jpan127, 2017-08-12:
Hahaha, oops.  I've ran shell scripts before on Windows.
Maybe it's just me but a directions section would be helpful.  Thanks anyways.

closed 2017-08-12 by jpan127

marcelstoer, 2017-08-12:
> directions section would be helpful

What exactly do you think is missing? Build instructions? Download instructions? Is the paragraph at the top of of the readme right below the screen shot not enough?

jpan127, 2017-08-12:
Well part of it is my unfamiliarity with terminology like binaries, so reading the section didn't pop out to me hey download the working application here!

I'm just saying a section titled "Setup" or "Installation".  Now I know though, but I'm sure there's going to be new people confused too.