marcelstoer/nodemcu-pyflasher #22: Feedback from the terminal window after Hard resetting.

AlesJaklic, 2017-08-12:
I receive no feedback in the terminal window after the message "Hard resetting..." after flashing the firmware and clearing the flash (Windows 7). Serial set to 115200. I expected to see: "Formatting the file system. Please wait..." or something. I simply wait for 2 to 3 minutes, so that this initial formatting has time to finish :) and then disconnect and reconnect the USB power to the Nodemcu board and everything is OK.  Am I doing something wrong or is this common on Windows?

marcelstoer, 2017-08-12:
Formatting the file system isn't part of the flashing process. I'll happen once you boot the device. You'll see that message if the device is connected to a serial console (e.g. ESPlorer) when it boots.

AlesJaklic, 2017-08-12:
But your software does Hard reset ... so I imagine the process of booting and Formatting the file system starts right away, there is no time to connect to some other serial console manually. If I disconnect the board right after the Hard resetting message I end up with unusable firmware, if I wait for a few minutes and then disconnect I get correct LUA prompt on LuaLoader. This my all sound as naïve superstition :), but it is highly reproducible for me.

marcelstoer, 2017-08-12:
Ah, I see...now I understand the issue.

It's actually not _my_ software that does the hard reset but esptool.py, I merely built the GUI and a standalone binary for it 😉 [It does that by default](https://github.com/espressif/esptool/wiki/Advanced-Options#reset-modes) but this could be changed.

A hard reset is like pressing the reset button on a devkit or otherwise restarting the device _in normal mode_ i.e. not bootloader for flashing. So, your observation is absolutely correct. After the hard reset NodeMCU starts and formats the file system if necessary. It has internal logic to determine whether formatting is necessary.

The problem is that esptool.py hung up on the serial connection after it triggered the hard reset. Hence, you won't see the NodeMCU boot messages.

I see how _not_ resetting could be useful but let me discuss this with the esptool.py folks first.

AlesJaklic, 2017-08-12:
I made a simple experiment using additional USB-to-Serial converter (COM11) at 115200 to capture the TX pin of the Nodemcu board during flashing and after the Hard resetting on (COM6). 

[image]

It shows clearly that the "Formatting file system. Please wait ..." is under way right after the Hard resetting ...  and if you disconnect the board during this, you get corrupted Nodemcu. It took less than a minute to format 4MB flash.  So a quick and dirty workaround is to notify the user, to wait a few minutes before disconnecting the board :)

marcelstoer, 2017-08-13:
Clever, it proves what I stated in the previous comment.

marcelstoer, 2017-08-16:
> I see how _not_ resetting could be useful but let me discuss this with the esptool.py folks first.

They agree with me -> I'll remove reset call.

commit 4d73e11d (2017-09-10): Stop triggering hard-reset after flash

Fixes #22
closed 2017-09-10 by marcelstoer