marcelstoer/nodemcu-pyflasher #24: WX error on build

djmason9, 2017-09-03:
I have pip install everything i can think of and i still get this error on my mac. I have python 2.7.10 and 3  and neither work

  File "nodemcu-pyflasher.py", line 3, in <module>
    import Main
  File "/Users/djmason9/Downloads/nodemcu-pyflasher-master/Main.py", line 3, in <module>
    import wx

codeorganic, 2017-09-04:
> Install wxPython 3.x (not 4.0 betas!)

Regardless, this still should be a valid bug to upgrade to `4.x` rather than continue on using `3.x`

marcelstoer, 2017-09-05:
It ain't a bug if a piece of software relies on stable dependencies rather than on beta and alpha releases. As said elsewhere this project _will_ eventually upgrade to wxPython 4 once a GA release becomes available.

closed 2017-09-05 by marcelstoer