marcelstoer/nodemcu-pyflasher #28: Requirements.txt / setup.py missing

rickardp, 2017-10-22:
Building this project is unnecessarily complex, since the standard python distutils is not used, and instead the build script just assumes everything installed (which means I have to guess and read the versions from README.md, etc).

Wouldn't it just be better to use the standard Python way of doing things (and publish releases on PyPi instead of custom builds)?

marcelstoer, 2017-10-22:
There's no doubt you know more about Python than I do. So, feedback is always welcome. Even better would be PRs helping me closing the gaps.

> Building this project is unnecessarily complex

Not for me (anymore) but potentially for you starting from scratch, I agree.

> Wouldn't it just be better to use the standard Python way of doing things (and publish releases on PyPi instead of custom builds)?

The primary goal is not to distribute a Python package but binaries for Windows, macOS and Linux eventually. In the future I might offload these tasks to TravisCI or similar.
However, that does not mean a PyPi presence is out of question. One is complementary to the other.

> Since you are using an old version of wxPython

I'm not. I stated here a couple of times that I would switch to wxPython 4.x as soon as there's a GA release rather than alpha/beta versions. I have reason to believe that the 4.x stream is not affected by pyinstaller/pyinstaller#2355.

murilopolese, 2018-11-22:
Hey, @marcelstoer !

I want to contribute to this project both improving it and solving bugs as it seems to be one of the best multi platform and open source GUI solutions for flashing ESP boards I've seen around (it does work with my ESP32 boards!!!)!

I'm also hitting a hard wall trying to setup the project both on my Linux (Ubuntu 18.04) and Windows 10 (virtual machine). 

I think it would be really beneficial for getting more contributions (and offload a bit the work from you) to document and make the setup/building process a bit more independent of host environment. I'd love to follow your suggestions on how to help but right now I am not even able to run it on my machines.

Could you give us what your current `python --version`, `pip --version` and `pip freeze` outputs?

marcelstoer, 2018-11-22:
Great news Murilo! You found the right issue 😜

https://github.com/marcelstoer/nodemcu-pyflasher#build-it-yourself is really minimal I admit.

I work on a Mac and macOS has a love-hate relationship with Python (being forever stuck on 2.7.x).

```
~ > python3 --version
Python 3.6.5
~ > python --version
Python 2.7.15
~ > pip --version
pip 18.0 from /usr/local/lib/python3.6/site-packages/pip (python 3.6)
~ > pip --freeze

Usage:   
  pip <command> [options]

no such option: --freeze
```

Looking forward to your contributions. I appreciate if you create issues here first before you start working on changes so we can discuss and exchange ideas before the implementation is done.

murilopolese, 2018-11-22:
I managed to make it run on my Ubuntu following specific instructions on `wxPython` download page. Windows is still not running with the following error:

```bash
$ python nodemcu-pyflasher.py
Traceback (most recent call last):
  File "nodemcu-pyflasher.py", line 3, in <module>
    import Main
  File "C:\Users\IEUser\Projects\nodemcu-pyflasher\Main.py", line 3, in <module>
    import wx
  File "C:\Users\IEUser\Projects\nodemcu-pyflasher\venv\lib\site-packages\wx\__init__.py", line 17, in <module>
    from wx.core import *
  File "C:\Users\IEUser\Projects\nodemcu-pyflasher\venv\lib\site-packages\wx\core.py", line 12, in <module>
    from ._core import *
ImportError: DLL load failed: The specified module could not be found.
```

I took the chance to create a `requirements.txt` and update the 'README.md` here: https://github.com/murilopolese/nodemcu-pyflasher/tree/feature/requirements.txt

marcelstoer, 2018-11-22:
Well done, thanks! Ready for PR?

The Windows issue may be related to #39 (or a dup thereof?).

JanusHL, 2018-11-23:
Hi Marcel, 
I friend og mine, test the program in windows 10 without problem, is an issue ny the moment in Windows 7 64x version of the tool...

murilopolese, 2018-11-23:
Hey, @marcelstoer! Here it is https://github.com/marcelstoer/nodemcu-pyflasher/pull/40

Regarding the Windows issue, it doesn't seems directly related but I need to further debug it. The DLL missing seems to be related with `wxPython` and it won't even mention which one is missing. I'll open a new issue about that to keep track of that (and maybe find out it's related with #39 ).

closed 2018-11-23 by marcelstoer