marcelstoer/nodemcu-pyflasher #29: Error after flashing an ESP32 based Lolin32 Lite: flash read err, 1000 ets_main.c 371

glederer007, 2017-10-28:
Hello,

I experiencing the following error after flashing an ESP32 based 4M flash LoLin32 Lite:

0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
....

If I use the Arduino IDE (which I think also uses the esptool.py), then I can program anything on the board, it will work. Of course in C, but I would like to use Lua or MicroPython.

Could you suggest what am I doing wrong?

Thanks,

marcelstoer, 2017-10-28:
Are you sure the flash params are correct? Have you tried with esptool.py directly? Arduino uses a different flasher: https://github.com/igrr/esptool-ck

glederer007, 2017-10-28:
Hello Marcel,

I tried with esptool.py directly just now. With it, I am able to flash the MicroPython binary, but the NodeMCU (Lua) binary is not going through... Well, it goes to a permanent loop.

Regards,

Gabor

glederer007, 2017-10-28:
Hello Marcel,

I tried again many times the LUA version with esptool.py without success. I compiled a basic ESP32 firmware for the Lolin32 based on https://github.com/nodemcu/nodemcu-firmware/tree/dev-esp32, 
 I flashed the board with it, but it still goes into loop at startup. Could you please help me with an experiment? Perhaps if you could compile a basic ESP32 firmware that you know it works, I would try to flash with that the board I have. 

Thanks in advance,

Gabor

marcelstoer, 2017-10-28:
If flashing with esptool.py doesn't work either then this issue is definitely unrelated to the NodeMCU PyFlasher app. I can currently devote very little time to NodeMCU and what you're asking for is individual user support, sorry.

closed 2017-10-28 by marcelstoer