marcelstoer/nodemcu-pyflasher #30: Error detecting flash size

limbo666, 2018-02-01:
I'm using the utility for long period with no issues on Wemos D1 modules, however there are some king of misbehavior with ESP-01 modules.
Form my latest test what I'm understanding is that the utility fails to detect correctly the flash size on the modules. 

Here how the process goes.
Step one: Flash the module. (it detects 1MB of flash size)
[image]
The procedure completes with no errors and at this point I think that the module is flashed correctly. 
But when I connect serial monitor there is no correct response from it. I getting the following: 
`ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 27780, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 2440, room 4 
tail 4
chksum 0xfa
load 0x3ffe8988, len 136, room 4 
tail 4
chksum 0x62
csum 0x62
csum err
ets_main.c 
` 
Trying to flash with a new firmware file (I just thought that the 1st one was over sized) the utility detects a different flash size
[image]
it never fisnished... 

So after resetting the module and try to re-flash the things gone extremely wrong... it seems that the esp module was bricked permanently. 
[image]
 
Is there any specific solution fro flashing ESP-01 modules with this GREAT UTILITY?

arendst, 2018-02-01:
Read the wiki and never use QIO

marcelstoer, 2018-02-02:
See https://github.com/marcelstoer/nodemcu-pyflasher#getting-help, this issue list is not a support forum.

Also note that this is "just" GUI for https://github.com/espressif/esptool. Hence, your issue is actually one with _that_ tool rather than with mine.

In any case, check whether your ESP-01 requires DIO rather than QIO.

closed 2018-02-02 by marcelstoer