marcelstoer/nodemcu-pyflasher #31: NodeMCU-PyFlasher-3.0 crashes on launch using macOS Sierra

phathak, 2018-02-14:
Hey,
mac user here using 10.12.5. I downloaded the latest releases today but I can't get to the GUI. It'll launch and then crash immediately. Have you seen this issue before?

Thank you.

marcelstoer, 2018-02-14:
Just verified on my usually dormant Mac Mini that it crashes on macOS Sierra. No idea why, the failure isn't even reported in the sys log here.

andrewoman, 2018-02-20:
10.12.6 as well. 
Service exited with abnormal code: 255

marcelstoer, 2018-02-20:
Yep, I guess all pre-"High Sierra" versions are affected because I build on Sierra.

MacGyverBeta, 2018-03-25:
Same here with 10.11.6. Console says:
"25.03.18 11:28:35,302 com.apple.xpc.launchd[1]: (com.frightanic.nodemcu-pyflasher.721632[3261]) Service exited with abnormal code: 255"

JieGH, 2018-05-25:
the app may unable to run in my mac,system version 10.12.6

masterkain, 2018-06-06:
```
[ruby-2.4.2] 22:30 ~ $ /Applications/NodeMCU-PyFlasher-3.0.app/Contents/MacOS/NodeMCU-PyFlasher
[1696] Error loading Python lib '/var/folders/4g/5_x3y4616cbbdysbv07hzf680000gn/T/_MEICDqR4S/Python': dlopen: dlopen(/var/folders/4g/5_x3y4616cbbdysbv07hzf680000gn/T/_MEICDqR4S/Python, 10): Symbol not found: _futimens
  Referenced from: /var/folders/4g/5_x3y4616cbbdysbv07hzf680000gn/T/_MEICDqR4S/Python (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib
 in /var/folders/4g/5_x3y4616cbbdysbv07hzf680000gn/T/_MEICDqR4S/Python

```

closed 2018-09-20 by phathak

roboterm, 2018-12-20:
any fixes for this issue?
why has it been closed?

marcelstoer, 2018-12-20:
Sierra is now two major OS versions behind. The next binary is likely going to be built on Mojave. If it happens to run on High Sierra, good, if not, too bad. I simply don't have the resources (neither macOS machines nor time) to build and test for OSs I don't personally use.