marcelstoer/nodemcu-pyflasher #32: error run latest code at ubuntu

johndoe71rus, 2018-03-04:
i run pyflasher before. After latest git pull i get error
` python ./nodemcu-pyflasher.py 
Traceback (most recent call last):
  File "./nodemcu-pyflasher.py", line 3, in <module>
    import Main
  File "/home/johndoe/Programs/esp8266/nodemcu-pyflasher/Main.py", line 4, in <module>
    import wx.adv
ImportError: No module named adv
`
lubuntu 14.04 
python-wxversion 3.0.1.1+dfsg-2~ubuntu14.04.1~ppa1
python-wxgtk3.0-dev 3.0.1.1+dfsg-2~ubuntu14.04.1~ppa1
python-wxgtk2.8 2.8.12.1+dfsg-2ubuntu2
python-wxgtk3.0 3.0.1.1+dfsg-2~ubuntu14.04.1~ppa1
installed

marcelstoer, 2018-03-04:
If you want to build it yourself or run Python directly then there's no way around following the [release notes](https://github.com/marcelstoer/nodemcu-pyflasher/releases) and read ["Build it yourself"](https://github.com/marcelstoer/nodemcu-pyflasher#build-it-yourself). You now need wxPython 4.x.

closed 2018-03-04 by marcelstoer

johndoe71rus, 2018-03-04:
i don't build. I just  run directly python ./nodemcu-pyflasher.py,

so i need old 2.2 version. How i can switch to 2.2 in my git copy?

marcelstoer, 2018-03-04:
In the releases section GitHub lists the Git tag for each release e.g. `v2.2`. You can simply switch your Git repository to any given tag with `git checkout v2.2`.