marcelstoer/nodemcu-pyflasher #34: nodemcu-pyflasher.py crashes on Ubuntu 14.04, cannot find wx module

BrendanHassett, 2018-04-11:
There seems to be a module dependency missing

./nodemcu-pyflasher.py
Traceback (most recent call last):
  File "./nodemcu-pyflasher.py", line 3, in <module>
    import Main
  File "/home/xxxxxxxxxxx/nodemcu-pyflasher/Main.py", line 3, in <module>
    import wx
ImportError: No module named wx

marcelstoer, 2018-04-11:
See https://github.com/marcelstoer/nodemcu-pyflasher#build-it-yourself and #28.

closed 2018-04-11 by marcelstoer

BrendanHassett, 2018-04-11:
After installing wx with the command
sudo apt-get install python-wxgtk2.8
a similar error states that module adv is missing

./nodemcu-pyflasher.py
Traceback (most recent call last):
  File "./nodemcu-pyflasher.py", line 3, in <module>
    import Main
  File "/home/xxxxxxxxxxxxx/nodemcu-pyflasher/Main.py", line 4, in <module>
    import wx.adv
ImportError: No module named adv

BrendanHassett, 2018-04-11:
Giving up.

The required version of wxpython depends on some specific version of GTK+. I just couldn't be bothered.
I agree with issue #28 that you need to document the build dependencies properly.

marcelstoer, 2018-04-11:
> I agree with issue #28 that you need to document the build dependencies properly.

Even if...due to the "Yes, we have Linux Wheels. Sort of." paragraph on https://wxpython.org/pages/downloads/ that wouldn't make lifes much easier for Linux-based developers.