marcelstoer/nodemcu-pyflasher #37: Release COM port when flash process is finished

amb-jarek, 2018-10-04:
**OS**
Windows 7

**NodeMCU-PyFlasher version**
3.0

**Description**
When I upload the firmware I'm not able to use Esplorer until I close the flasher. It looks that COM port it is still locked.

**Expected behavior**
COM port can be used by any other app as soon as Flasher finish his job without closing it.

marcelstoer, 2018-10-21:
The PyFlasher doesn't do a hard reset after flashing on purpose. This was changed about a year ago after discussions in #22. Please see if the reasons stated there makes sense to you as well.

amb-jarek, 2018-10-21:
I'm not sure if this is my case. When I flash new firmware using `nodemcu-pyflasher` the port is locked even if I restart my ESP module or even power the ESP off and on. 

The important thing which I forgot to mention is that I'm not using Nodemcu Board but regular ESP-07 module connected through USB to TTL Serial adapter (FTDI).

The problem here is that not ESP module itself but USB serial adapter is locked and can't be opened by any app (ESPlorer, putty, or any other terminal client) untill `nodemcu-pyflasher` is closed.

marcelstoer, 2018-10-22:
I'm afraid there's nothing I can do here, sorry. I'll close it for a number of reasons:
- too many variables
  - OS
  - driver
  - serial adapter
- none of my test devices ever showed this behavior
- the PyFlasher is just a GUI for esptool.py, maybe search for [open issues](https://github.com/espressif/esptool/issues) or help there

closed 2018-10-22 by marcelstoer

amb-jarek, 2018-10-22:
@marcelstoer 
I'm using newest version of esptool (v2.5.1) without any issues, but it looks that problem maybe located there indeed. As I see you are using `esptool v2.2` in the flasher, right?

I just checked the `esptool` changelog and I found interesting entry in v2.4.0 which is:
```
Explicitly close the serial port at end of main() function (allows calling
esptool.main() from another Python program for basic scripting.)
```
so it could be the reason that flasher locks the serial port while esptool (newest) executed directly not.

marcelstoer, 2018-10-22:
Right, [currently still on 2.2.1](https://github.com/marcelstoer/nodemcu-pyflasher/blob/master/esptool.py#L36). I had wanted to upgrade for quite a while as v2.4 brought auto-serial port detection (which means I could potentially remove a GUI field).

In a fork I maintain for my side business I already upgraded to 2.5 - [was straight forward](https://github.com/ThingPulse/app-fairy/commit/18cc721cbfd1a824c2dfc678aa3933bdaa3dc8a9).

I created #38 for that.