marcelstoer/nodemcu-pyflasher #41: Missing DLL when importing `wxPython`

murilopolese, 2018-11-23:
After installing the dependencies on Windows 10 ([MSEdge virtual box image](https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/)) I get the following error trying to run the application:

```
$ python nodemcu-pyflasher.py
Traceback (most recent call last):
  File "nodemcu-pyflasher.py", line 3, in <module>
    import Main
  File "C:\Users\IEUser\Projects\nodemcu-pyflasher\Main.py", line 3, in <module>
    import wx
  File "C:\Users\IEUser\Projects\nodemcu-pyflasher\venv\lib\site-packages\wx\__init__.py", line 17, in <module>
    from wx.core import *
  File "C:\Users\IEUser\Projects\nodemcu-pyflasher\venv\lib\site-packages\wx\core.py", line 12, in <module>
    from ._core import *
ImportError: DLL load failed: The specified module could not be found.
```

I am not sure if this has anything to do with `nodemcu-pyflasher` itself but perhaps `wxPython` requires a specific version of the package, Python, pip, and maybe some setup on the host machine.

cs8898, 2019-06-08:
Did you install wxPython on your Win machine?

https://wxpython.org/pages/downloads/index.html

Seems not related to the other issue, the other one seems to be more python base specific

marcelstoer, 2019-10-06:
This is a wxPython issue.

closed 2019-10-06 by marcelstoer