marcelstoer/nodemcu-pyflasher #44: UI is ambiguous about which of the 4 files to specify for the "NodeMCU Firmware"

NewburyCSAF, 2019-01-10:
Thanks for the tool! However, There doesn't seem to be any "instructions" or indications about which of the 4 bin files I should be selecting in the Browse window for NodeMCU Firmware selection.

Yes, I agree that there should be some indication of completion. :-D

marcelstoer, 2019-01-10:
> However, There doesn't seem to be any "instructions" or indications about which of the 4 bin files I should be selecting in the Browse window for NodeMCU Firmware selection.

The tool doesn't know the semantics of the folder you point the file browser to. It may contain 1 bin (e.g. the one you downloaded from somewhere), there may be 2 if you built NodeMCU locally or there may be a ton if you keep older versions. So, it's the context in which you work that defines which bin to select.

> Yes, I agree that there should be some indication of completion. :-D

There is. The console output says that everything is done and tells you what to do next.

closed 2019-01-10 by marcelstoer

NewburyCSAF, 2019-01-14:
> Arana Fireheart MSEE
> Asst. Prof. Computer Science
> Newbury College
> West Hall 201, 129 Fisher Ave., Brookline, MA 02445
> C: 781-760-4135
> E: arana.fireheart@newbury.edu

Marcel,

    Correct m if I am wrong, but doesn’t MicroPython require 4 bin files to be loaded in order to be complete? What I am saying, is that maybe the UI should allow the selection of those 4 separately, that would indicate to a new user that there are multiple files required by MicroPython, even if you just chose to load one.

    I’ll try it again and looks for the completion notice, but I used it a few thousand mes and don’t recall seeing any notification of completion.

marcelstoer, 2019-01-14:
Ahhh, I think you are mistaking this tool for something that it isn't.

> doesn’t MicroPython require 4 bin files...

The NodeMCU PyFlasher is intended to be used for the NodeMCU firmware (Lua). It flashes a single binary to address 0x00000 so you _can_ actually use it to flash other "firmware" as well. I sometimes use it to flash Arduino binaries to an ESP8266.

I've only briefly used MicroPython on the very first WiPy two or so years ago. I don't remember how you would upgrade the firmware.

NewburyCSAF, 2019-01-15:
Marcel,

    THAT explains A LOT! 😁 

    Maybe I’ll pick up the mantle and build the one I was looking for. It would be using the same underlying esptool S/W...

    Any advice in that endeavor?

___________________
Arana Fireheart MSEE
Asst. Prof. Computer Science
Newbury College
West Hall 201, 129 Fisher Ave., Brookline, MA 02445
C: 781-760-4135
E: arana.fireheart@newbury.edu