marcelstoer/nodemcu-pyflasher #46: This program needs access to the screen. Please run with a Framework build of python, and only when you are logged in on the main display of your Mac.

wolfiex, 2019-04-07:


marcelstoer, 2019-05-24:
Some information on this WxPython issue here: https://stackoverflow.com/questions/39989975/wx-cant-run-on-mac

sshah254, 2020-06-15:
Hi,

I ran into this issue too.

When I was building it on my own from the command line, for example,

python3 -m PyInstaller -n NodeMCU-PyFlasher nodeycu-pyflasher.py

then if I run the executable from the dist directory, then I'll get the above error.

But if I build it with

python3 -m PyInstaller -w -n NodeMCU-PyFlasher nodeycu-pyflasher.py

(notice -w option for --windowed)

then I won't get the above error.

There were more issues but that I have listed in issue #61 (https://github.com/marcelstoer/nodemcu-pyflasher/issues/61)

hth,

SS

marcelstoer, 2021-02-18:
Even mentioned at the wxPython wiki at https://wiki.wxpython.org/wxPythonVirtualenvOnMac.

> (notice -w option for --windowed)

Future versions will be built with `windowed` enabled.

closed 2021-02-18 by marcelstoer