marcelstoer/nodemcu-pyflasher #49: [FeatureRequest] Add options to select partition offset

riedel, 2019-06-07:
This is the default setup for a heltec device:
`esptool.py --chip esp32 --port COM8  --baud 921600 --before default_reset --after hard_reset write_flash -z 0x1000 bootloader.bin 0x8000 partitions.bin 0xe000 boot_app0.bin  0x10000 wikit_firmware_*`

writing to 0x0000 will erase the bootloader...

marcelstoer, 2019-06-07:
I won't add that for the reasons outlined at https://github.com/marcelstoer/nodemcu-pyflasher#motivation. 

This app should allow you to flash a single _combined_ binary with as few options as possible. In the NodeMCU world we use [`srec_cat`](http://srecord.sourceforge.net/man/man1/srec_cat.html) to combine and pad (where necessary) individual binaries. Examples:
- https://github.com/marcelstoer/docker-nodemcu-build/blob/master/build-esp32#L70
- https://github.com/marcelstoer/docker-nodemcu-build/blob/master/build-esp8266#L93
- https://github.com/nodemcu/nodemcu-firmware/blob/master/.travis.yml#L21

closed 2019-06-07 by marcelstoer

riedel, 2019-06-08:
Thank you for the helpful links! `srec_cat` might be  indeed a better option in our case.