marcelstoer/nodemcu-pyflasher #50: Not working on specific computer

adizag, 2019-06-21:
Hi

Using my laptop (win10 64, com 9 recognized) trying to flash. the log shows only the 1st line of the "command:" and doesn't proceeds to the flashing operation. One line written only, no flashing. Tried any possible option including open as administrator, installing  ftdi drivers... Nothing.

On another old and sluggish  laptop, the tool is working just great.

I'd appreciate your assistance!

Ovadex, 2019-06-23:
I was having the same problem on my thinkpad. On mine it was shown in device manager as "ftdi device (com 9)". I got it working by uninstalling the ftdi device (and selecting the delete software option) and then re-plugging the adapter so it reinstalls.

After it installed it was listed as "usb serial port (com 9)" and it worked properly.

Hope this helps!

adizag, 2019-06-23:
Just tried (including SW). didn't work... :(

____________________________________________
"יש המטביעים את צרותיהם - ויש המלמדים אותן לשחות..."
"אין מקום שהוא רחוק מדי"

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
בלי
וירוסים. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

marcelstoer, 2019-10-06:
https://github.com/marcelstoer/nodemcu-pyflasher#getting-help

closed 2019-10-06 by marcelstoer