marcelstoer/nodemcu-pyflasher #51: get memory size

ingoiot, 2019-07-21:
would it be possible to add a Button to get memory size (or as much ifo about the chip)
before flashing?

--flash_size=detect
--port /dev/ttyUSB0 flash_id

for example?

marcelstoer, 2019-07-21:
You can only get this as a side effect of running other commands:

```
~/Data/NodeMCU/nodemcu-firmware (master) > esptool.py --port=/dev/cu.SLAB_USBtoUART flash_id
esptool.py v2.7
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:d9:5f:0b
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4018
Detected flash size: 16MB
Hard resetting via RTS pin...
~/Data/NodeMCU/nodemcu-firmware (master) > esptool.py --port=/dev/cu.SLAB_USBtoUART read_mac
esptool.py v2.7
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:d9:5f:0b
Uploading stub...
Running stub...
Stub running...
MAC: 5c:cf:7f:d9:5f:0b
Hard resetting via RTS pin...
~/Data/NodeMCU/nodemcu-firmware (master) > esptool.py --port=/dev/cu.SLAB_USBtoUART chip_id
esptool.py v2.7
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:d9:5f:0b
Uploading stub...
Running stub...
Stub running...
Chip ID: 0x00d95f0b
Hard resetting via RTS pin...
```

ingoiot, 2019-07-21:
from the esptool.py site:

_[Read built-in MAC address: read_mac

esptool.py read_mac

Read SPI flash id: flash_id

esptool.py flash_id

Example output:

Manufacturer: e0
Device: 4016
Detected flash size: 4MB

Refer to flashrom source code for flash chip manufacturer name and part number.](url)_

looked like it should work.

closed 2019-10-06 by marcelstoer