marcelstoer/nodemcu-pyflasher #53: Add backup option

rt400, 2019-08-22:
hi..
there is a way to add a backup option ?

artmg, 2019-10-05:
According to [these instructions](https://blog.hobbytronics.pk/sonoff-original-firmware-backup-restore/) `esptool` has a read_flash instruction that could be used for a backup option

```
# back up 1MB image
esptool.py --port COM5 read_flash 0x00000 0x100000 image1M.bin
# back up 4MB image
esptool.py --port COM5 read_flash 0x00000 0x400000 image4M.bin
```

so it should be possible, in `main.py` to add a second click button called `backup First` and get this to run one or both of the commands above. You might presume the location to save to from the location of the firmware file chosen with the `Browse` button.

marcelstoer, 2019-10-06:
This was designed as a single-purpuse - simplicity is key. There has to be a lot of convincing demand from the community until I consider adding more options and features. I'll keep this open for now, though.

kerozen54, 2019-10-31:
Hi all.
Yes, this will be a great feature for this tool.
It is a question of security first. Backup the actual firmware before flashing a new one seems to be a necessary option.

IanAdd, 2020-05-07:
Yes please, backup would be a safer way forward in case of issues with an unproven device.