marcelstoer/nodemcu-pyflasher #60: ESP32 keeps resetting after flashing with NodeMCU-PyFlasher-4.0-x64

JugensVan, 2020-03-17:
When i program my esp32 with a simple flash example  using NodeMCU-PyFlasher-4.0-x64
it keeps resetting.
Can anybody help.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

Manuel-Sensate, 2020-04-22:
I stumbled over a similar issue - the problem you might have run into is if you are just trying to flash a "barebone" app image binary. you need to prepare the binary (i.e. combine the bootloader, apploader, app and partition binary files to one binary) - then it works fine.

marcelstoer, 2020-04-22:
> combine the bootloader, apploader, app and partition binary files to one binary

That's correct.

As for `flash read err, 1000` there is some information at https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#early-flash-read-error.

JugensVan, 2020-04-22:
> I stumbled over a similar issue - the problem you might have run into is if you are just trying to flash a "barebone" app image binary. you need to prepare the binary (i.e. combine the bootloader, apploader, app and partition binary files to one binary) - then it works fine.

Thanks That makes sense. Just not to sure how to do that. I compile using Platform IO. The only bin files i can find is firmware.bin and one called partitions.bin  Im new to ESP32. Im use to using HEX files for programming.

JugensVan, 2020-04-22:
> I stumbled over a similar issue - the problem you might have run into is if you are just trying to flash a "barebone" app image binary. you need to prepare the binary (i.e. combine the bootloader, apploader, app and partition binary files to one binary) - then it works fine.

Thanks that makes sense. Im new to ESP32. i compile using PlatformIO. Not  sure how to combine .bin files. The only bin files i can find in my project are firmware.bin and partitions.bin. I'm used to using .HEX files in batch programming.  any help will be apresiated.

Manuel-Sensate, 2020-04-22:
It is indeed a bit tricky.. lots of caveats.. I used
https://github.com/vtunr/esp32_binary_merger - the required other binaries you find in the Arduino SDK for ESP32.

marcelstoer, 2020-04-22:
> https://github.com/vtunr/esp32_binary_merger 

That's one option (one I didn't know existed). The other is using `srec_cat` as I do in the NodeMCU ESP32 build at https://github.com/marcelstoer/docker-nodemcu-build/blob/master/build-esp32#L85.

JugensVan, 2020-04-22:
> It is indeed a bit tricky.. lots of caveats.. I used
> https://github.com/vtunr/esp32_binary_merger - the required other binaries you find in the Arduino SDK for ESP32.

Thank you. Will try that.

JugensVan, 2020-04-22:
> > https://github.com/vtunr/esp32_binary_merger
> 
> That's one option (one I didn't know existed). The other is using `srec_cat` as I do in the NodeMCU ESP32 build at https://github.com/marcelstoer/docker-nodemcu-build/blob/master/build-esp32#L85.

Thanks now i know how to sort it.

uzi18, 2020-04-22:
have prepared python script to hook after platformio build ends - it create genuine flash image with all needed stuff

marcelstoer, 2020-04-22:
@uzi18 I haven't tested it yet but that's a great idea! Once I hear from others who successfully use this I will documented it here in the README.

uzi18, 2020-04-22:
maybe @JugensVan will try to test
basically it creates combined file firmware-factory.bin in build dir
you can flash it at 0x0 and from this moment just use ota update with or only reflash firmware.bin =)

JugensVan, 2020-04-23:
Sounds like a good idea. Problem is im new to this platform. and not to sure how to do that. Will ask one of my colleagues to try it. He is ofay with python script .

uzi18, 2020-04-23:
just copy this file to project dir and add extra line into env declaration in platformio.ini
you don't need to know python =)

uzi18, 2020-04-23:
like here: https://github.com/letscontrolit/ESPEasy/blob/mega/platformio_esp32_envs.ini#L18

JugensVan, 2020-04-23:
> just copy this file to project dir and add extra line into env declaration in platformio.ini
> you don't need to know python =)

I just copied the script to my project directory and added "extra_scripts  = post:post_esp32.py" to platformIO.ini
it made a "firmware-factory.bin" 
i flash the firmware using  NodeMCU-PyFlasher-4.0-x64 
It all seems to work fine.
I thank you for all your help. 
I'm sure more people will find this useful if you need to send a bin file to someone to upgrade firmware in the field.

closed 2020-07-31 by marcelstoer