marcelstoer/nodemcu-pyflasher #61: Building it myself failes (Mac)

Manuel-Sensate, 2020-04-22:
I am trying to build a release myself (actually my target is to get a notarized build that works in Catalina), but I run into this.

`88 INFO: PyInstaller: 3.6
88 INFO: Python: 3.7.7
96 INFO: Platform: Darwin-19.4.0-x86_64-i386-64bit
96 DEBUG: Testing for UPX ...
108 INFO: UPX is available.
109 DEBUG: script: nodemcu-pyflasher.py
109 INFO: Extending PYTHONPATH with paths
['/Users/*user*/Downloads/nodemcu-pyflasher-master']
110 INFO: checking Analysis
118 INFO: Appending 'datas' from .spec
122 INFO: checking PYZ
125 INFO: checking PKG
127 INFO: Building because toc changed
127 INFO: Building PKG (CArchive) PKG-00.pkg
537 DEBUG: Compiling /usr/local/Cellar/pyinstaller/3.6/libexec/lib/python3.7/site-packages/PyInstaller/loader/pyiboot01_bootstrap.py
538 DEBUG: Compiling /usr/local/Cellar/pyinstaller/3.6/libexec/lib/python3.7/site-packages/PyInstaller/loader/pyiboot01_bootstrap.py
539 DEBUG: Compiling /Users/*user*/Downloads/nodemcu-pyflasher-master/nodemcu-pyflasher.py
539 DEBUG: Compiling /Users/*user*/Downloads/nodemcu-pyflasher-master/nodemcu-pyflasher.py
2028 INFO: Building PKG (CArchive) PKG-00.pkg completed successfully.
2031 INFO: Bootloader /usr/local/Cellar/pyinstaller/3.6/libexec/lib/python3.7/site-packages/PyInstaller/bootloader/Darwin-64bit/runw
2031 INFO: checking EXE
2033 INFO: Building because toc changed
2033 INFO: Building EXE from EXE-00.toc
2033 INFO: Appending archive to EXE /Users/manuel.fegerl/Downloads/nodemcu-pyflasher-master/dist/NodeMCU-PyFlasher
2037 INFO: Fixing EXE for code signing /Users/manuel.fegerl/Downloads/nodemcu-pyflasher-master/dist/NodeMCU-PyFlasher
2040 INFO: Building EXE from EXE-00.toc completed successfully.
2045 INFO: checking BUNDLE
On your own risk, you can use the option `--noconfirm` to get rid of this question.
2046 INFO: Removing dir /Users/*user*/Downloads/nodemcu-pyflasher-master/dist/NodeMCU-PyFlasher-4.0.app
2047 INFO: Building BUNDLE BUNDLE-00.toc
2056 INFO: moving BUNDLE data files to Resource directory
Not enough arguments. Invoke with --help for help.
mv: rename NodeMCU-PyFlasher-4.0 0.0.0.dmg to dist/NodeMCU-PyFlasher-4.0.dmg: No such file or directory`

My approach: I first installed the environment + requirements like you described in the readme (seems to work), but this happens when i try the build.sh script.

Any ideas?

Manuel-Sensate, 2020-04-24:
Any hints on what I am doing wrong would be appreciated.

aLoneStrider, 2020-05-11:
It seems you did install python wx on  a different python version. Are you able to double check the python version and wx?

sshah254, 2020-06-15:
Hi,

I had a similar problem on my macOS Catalina 10.15.5.

I did the following to make it work (all the commands were run in terminal from the directory that contained the source code, in my case, from /Users/ss/Desktop/ESP8266/nodemcu-pyflasher):

1.  I am using the 4.0 master branch from GitHub from this weekend (06/14/2020).  I edited the Main.py file and replaced lines 222 and 301 as follows:

222 -        #serial_boxsizer.Add(reload_button, 0, wx.ALIGN_RIGHT, 20)
replaced -        serial_boxsizer.Add(reload_button, 0)
301 -         #flashmode_label_boxsizer.Add(icon, 0, wx.ALIGN_RIGHT, 20)
replaced -        flashmode_label_boxsizer.Add(icon, 0)

2.  Manually built it using
python3 -m PyInstaller -w -n NodeMCU-PyFlasher nodemcu-pyflasher.py
(python -m PyInstaller -w -n .... OR pyinstaller -w -n ... should work fine too, but ymmv).
3.  Created two directories that were causing the app to fail (dist/NodeMCU-PyFlasher.app/Contents/MacOS/Tk and dist/NodeMCU-PyFlasher.app/Contents/MacOS/tcl).  You may not need these, am not sure if this is an issue with my setup or not.
4.  Generated my own "NodeMCU-PyFlasher" code sign certificate in KeyChain.
5.  Signed all the binaries using that certificate.
find dist/NodeMCU-PyFlasher.app -perm +111 -type f -or -type l | xargs -I file codesign -f -s NodeMCU-PyFlasher --deep file

After this, my NodeMCU-PyFlasher app launches just fine.

SS

closed 2020-07-31 by marcelstoer