marcelstoer/nodemcu-pyflasher #62: Problem for detect serial ports

GSmurf, 2020-04-29:
Hi,

When I launch nodemcu I reload serial device list but no one appear. And of course when I click Flash after select my firmware and all options I got this message :

> Command: esptool.py --baud 115200 --after no_reset write_flash --flash_mode dout 0x00000 Mypath\sonoff\tasmota-FR.bin --erase-all
> 
> esptool.py v2.6
> Found 0 serial ports

Have you an idea about the problem ?

Ty for your help

aLoneStrider, 2020-05-11:
Hi @GSmurf ,
It seems nothing detected on your ports. it might been caused by [driver](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers) and permission on ports or maybe a power only USB cable.

GSmurf, 2020-05-17:
Hi ! Ty @aLoneStrider effectivly it was a pb with USB cable XD

> Use a USB data cable - Some USB cables are for charging only and do not connect the data lines needed to load the firmware onto the device.

TY for your help

closed 2020-05-17 by GSmurf