marcelstoer/nodemcu-pyflasher #70: Setting partitions

wizwik, 2020-09-03:
I have looked around but cannot find where to set the partitions.  I am trying to load an Arduino produce binary file and it is loading to address 0x00000. I Believe it should be loaded to 0x10000 but I cannot find where to do this. Here is my log showing the load to 0x00000.
Wrote 207856 bytes (105418 compressed) at 0x00000000 in 9.3 seconds (effective 178.7 kbit/s)...

Regards

wizwik

marcelstoer, 2020-09-03:
You cannot set partitions. This expects a single binary including app, bootloader and partitions all placed at the correct (memory) addresses.

closed 2020-09-03 by marcelstoer

wizwik, 2020-09-04:
> Links:

Hi Marecel:

Thanks very much. I was hoping to load a Bin file to 0x10000 but I see 
that is not possible;

Regards

John