marcelstoer/nodemcu-pyflasher #76: GUI multilanguage

bovirus, 2021-04-24:
@marcelstoer 

Could be please condider to have a GUI multilanguage?

Thanks.

bovirus, 2021-05-11:
@marcelstoer

Could you please answer to the request?

marcelstoer, 2021-05-12:
> Could be please consider to have a GUI multilanguage?

If someone is willing to contribute this feature I will certainly consider it. However, I personally do not see much benefit in having a localized GUI as the content of the UI-console will remain in English anyhow (produced by esptool.py). What is the use case you have in mind?

bovirus, 2021-05-12:
@marcelstoer

For me flash program is very dangerous program if the users doens't know/understand the meaning of some functions.

In this case a localized GUI about flashing can help.