marcelstoer/nodemcu-pyflasher #77: Unexpected quitting after pressing the flash button on Apple M1 machine (Big Sur)

tommyjtl, 2021-06-08:
See below screenshot and warning:

[image]

```
Process:               NodeMCU PyFlasher [30673]
Path:                  /Applications/NodeMCU PyFlasher.app/Contents/MacOS/NodeMCU PyFlasher
Identifier:            com.frightanic.nodemcu-pyflasher
Version:               5.0.0 (???)
Code Type:             X86-64 (Translated)
Parent Process:        NodeMCU PyFlasher [30672]
Responsible:           NodeMCU PyFlasher [30672]
User ID:               501

Date/Time:             2021-06-08 10:53:51.521 +0800
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Anonymous UUID:        E79777C0-C33C-6D97-7F55-C6AFDD85B90C

Sleep/Wake UUID:       8EE7DCB0-EBD9-44C3-B811-2E91CB9F6410

Time Awake Since Boot: 88000 seconds
Time Since Wake:       1500 seconds

System Integrity Protection: enabled

Crashed Thread:        8

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000100001280
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [30673]

VM Regions Near 0x100001280:
--> 
    __TEXT                      100311000-100319000    [   32K] r-x/r-x SM=COW  /Applications/NodeMCU PyFlasher.app/Contents/MacOS/NodeMCU PyFlasher

Application Specific Information:
objc_msgSend() selector name: layoutManager

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_malloc.dylib        	0x00007fff20172829 szone_malloc_should_clear + 324
1   libsystem_malloc.dylib        	0x00007fff2018c09e _malloc_zone_calloc + 59
2   libobjc.A.dylib               	0x00007fff201f2d37 _objc_rootAllocWithZone + 50
3   com.apple.Foundation          	0x00007fff2110713b -[NSConcreteMutableAttributedString initWithAttributedString:] + 580
4   com.apple.Foundation          	0x00007fff2112f65c -[NSConcreteMutableAttributedString initWithString:attributes:] + 122
5   com.apple.AppKit              	0x00007fff22d19db5 -[NSTextView(NSTextCheckingClient) annotatedSubstringForProposedRange:actualRange:] + 2061
6   com.apple.AppKit              	0x00007fff22d1936d -[NSTextCheckingController annotatedSubstringForProposedRange:wrap:completionHandler:failureHandler:] + 436
7   com.apple.AppKit              	0x00007fff22d191b0 -[NSTextCheckingController annotatedSubstringForProposedRange:completionHandler:] + 84
8   com.apple.AppKit              	0x00007fff22d1b1fc -[NSTextCheckingController annotatedSubstringForSelectedRangeWithCompletionHandler:] + 77
9   com.apple.AppKit              	0x00007fff22d1b1a3 -[NSTextCheckingController _checkBubblesAfterMovementFromRange:] + 124
10  com.apple.AppKit              	0x00007fff22d190bb -[NSTextCheckingController didChangeSelectionFromRange:] + 121
11  com.apple.AppKit              	0x00007fff22d17240 -[NSTextView(NSSharing) setSelectedRanges:affinity:stillSelecting:] + 3002
12  libwx_osx_cocoau_core-3.1.5.0.0.dylib	0x000000010d23a82e non-virtual thunk to wxNSTextViewControl::SetSelection(long, long) + 142
13  libwx_osx_cocoau_core-3.1.5.0.0.dylib	0x000000010d3616e6 wxTextEntryBase::AppendText(wxString const&) + 22
14  _core.cpython-39-darwin.so    	0x000000010b978671 meth_wxTextEntry_AppendText(_object*, _object*, _object*) + 161
15  Python                        	0x000000010a834c84 cfunction_call + 52
16  Python                        	0x000000010a7eac19 _PyObject_Call + 125
17  Python                        	0x000000010a8e09a9 _PyEval_EvalFrameDefault + 45894
18  Python                        	0x000000010a8d451b _PyEval_EvalCode + 403
19  Python                        	0x000000010a7eaea4 _PyFunction_Vectorcall + 376
20  _core.cpython-39-darwin.so    	0x000000010b6d5d21 wxPyCallback::EventThunker(wxEvent&) + 289
21  libwx_baseu-3.1.5.0.0.dylib   	0x000000010e255f4d wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 317
22  libwx_baseu-3.1.5.0.0.dylib   	0x000000010e255c5b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
23  libwx_baseu-3.1.5.0.0.dylib   	0x000000010e255b14 wxEvtHandler::ProcessEvent(wxEvent&) + 100
24  _core.cpython-39-darwin.so    	0x000000010b89e1eb sipwxPyApp::ProcessEvent(wxEvent&) + 107
25  libwx_baseu-3.1.5.0.0.dylib   	0x000000010e255763 wxEvtHandler::ProcessPendingEvents() + 323
[... 555 lines cut ...]
__UNICODE                          588K        1 
mapped file                        4.1G      570 
shared memory                      792K       14 
unshared pmap                     10.5M        6 
===========                     =======  ======= 
TOTAL                              6.6G     2315 
TOTAL, minus reserved VM space     5.6G     2315 

Model: MacBookPro17,1, BootROM 6723.120.36, proc 8:4:4 processors, 16 GB, SMC 
Graphics: kHW_AppleM1Item, Apple M1, spdisplays_builtin
Memory Module: LPDDR4
AirPort: spairport_wireless_card_type_airport_extreme, wl0: Apr 28 2021 19:53:33 version 18.50.40.7.7.8.119 FWID 01-e6894c69
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: USB 3.1 Bus
USB Device: USB3.1 Hub
USB Device: USB 10/100/1000 LAN
USB Device: USB2.1 Hub
USB Device: USB 2.0 Hub
USB Device: BillBoard Device
USB Device: CP2102 USB to UART Bridge Controller
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

marcelstoer, 2021-06-08:
PyInstaller only just added M1 support a couple of days ago: https://github.com/pyinstaller/pyinstaller/pull/5581. I guess I should mention this on the releases page.

L1cardo, 2021-11-25:
I have tested on macOS Monterey 12.0.1 (21A559) on my M1 Mac mini , and all things went well.
Since PyInstaller has released `v4.7`， maybe you can consider to update nodemcu-pyflasher for better performance.
[image]

closed 2022-01-18 by marcelstoer

fortiko, 2022-07-19:
@marcelstoer I checked out the latest version from git yesterday, and had to make 2 small changes to requirements.txt:

- Use latest v4 release of PyInstaller
- Use latest v3 release of esptool, at the moment any later version causes recursion errors, see [https://pythontechworld.com/issue/espressif/esptool/749]

After those changes, building on M1 Pro works out of the box.

```
diff --git a/requirements.txt b/requirements.txt
index cc04c8c..0e23417 100644
--- a/requirements.txt
+++ b/requirements.txt
@@ -1,6 +1,6 @@
-esptool>=3.0
+esptool~=3.3.1
 pyserial~=3.5
 wxPython~=4.1.1
-PyInstaller~=4.8
+PyInstaller~=4.10
 httplib2>=0.18.1
 pyinstaller-versionfile>=2.0.0
```

commit 2fbc4031 (2022-01-18): Bump PyInstaller to 4.8

Fixes #77