marcelstoer/nodemcu-pyflasher #78: NodeMCU-PyFlasher.exe (v5.0.0) Not Working on Windows 10

Gosokan, 2021-06-09:
Currently using Windows 10 64-bit version 20H2 (OS Build 19042.867)
NodeMCU-PyFlasher.exe v5.0.0 does not display anything when this program is executed.

starioni, 2021-06-18:
Same to me. running the .exe doesn't do anything

rt400, 2021-06-25:
also same to me. running and closing after couple seconds.

Fevax, 2021-09-04:
Same issue here

exxoff, 2021-10-21:
Same. Process starts but stops immediately. v.4 x64 works.

marcelstoer, 2021-11-22:
I'd appreciate if everybody following along here could spend 2min testing this version I built on yet another windows: https://transfer.sh/Ha0Wj8/NodeMCU-PyFlasher.exe

Thanks for your feedback.

rt400, 2021-11-23:
link not working

marcelstoer, 2021-11-23:
Thanks for testing. Works for me but I published it to an alternative location (my own domain): https://frightanic.com/misc/NodeMCU-PyFlasher.exe.

rt400, 2021-11-23:
still not working...

Gosokan, 2021-11-24:
I also tried it. Still not working

wkr1337, 2022-01-16:
Same issue, running the .exe doesn't do anything.
Links do not work.

rt400, 2022-01-18:
@marcelstoer HI, i think i fix the problem by add locate to your code..
import locate`
locale.setlocale(locale.LC_ALL,'C')
`
its work for me... plz check it and let me know

marcelstoer, 2022-01-18:
Both versions side-by-side: 

- https://frightanic.com/misc/NodeMCU-PyFlasher-without-locale.exe
- https://frightanic.com/misc/NodeMCU-PyFlasher.exe

faizakbar18, 2022-01-18:
not working for me, still does not display anything

rt400, 2022-01-18:
> not working for me, still does not display anything

https://github.com/rt400/Tasmota-Config-Backup/raw/master/NodeMCU-PyFlasher.exe

Please try this and let me know if its work...

rt400, 2022-01-18:
> @rt400 thanks, I can definitely try that. I for one won't notice any difference (as the app works just fine on _my_ Windows).
> 
> How did you arrive at this conclusion? How did you debug this?

Well I started playing with your code, I tried to create the BUILD myself but it did not work.
I finally tried running it through the CONSOLE and saw that it threw an error on something belonging to LOCATE. After an in-depth Google search I saw a solution that suggests adding these lines and this should solve the problem. By the way this is a known issue of wxPython and it has not yet been resolved.
I added these lines and created a new BUILD and it just opens and runs.
Where did you Add those lines ? cause what you send not open for me..

marcelstoer, 2022-01-18:
@rt400 Thanks for the hints but your comment left me a bit puzzled.

In your earlier comment at https://github.com/marcelstoer/nodemcu-pyflasher/issues/78#issuecomment-950192042 you said that you had run it through the console and it didn't give any hints? How apparently

> it threw an error on something belonging to LOCATE

Btw, I used `import locale` not `locate` as the latter does not exist.

I asked my favorite search engine about this "locale + wxPython" issue and found a couple of discussions which I'm linking below for reference:

- https://discuss.wxpython.org/t/windows-locale-error-wx-core-wxassertionerror/35714
- https://discuss.wxpython.org/t/wxpython4-1-1-python3-8-locale-wxassertionerror/35168
- https://discuss.wxpython.org/t/what-is-wxpython-doing-to-the-locale-to-makes-pandas-crash/34606

One other thing... above you both said 
> I tried to create the BUILD myself but it did not work. 
> ...
> I added these lines and created a new BUILD

I am assuming (since you uploaded a `.exe`) you eventually did manage to fix this.

rt400, 2022-01-18:
@marcelstoer 
in comment 78 I run only your builds file (.EXE) that you upload and one of them should open the console, also run it the EXE from cmd.
Now i download your build and its work... Finally !!!!
Greate Job ! I used alot with your pyFlasher.

exxoff, 2022-01-18:
@marcelstoer I can confirm, it works for me too now.

Good work!

 / Per

marcelstoer, 2022-01-18:
Thanks folks for sticking around and for providing valuable feedback.

**tl;dr**
Until I create a new release please use https://frightanic.com/misc/NodeMCU-PyFlasher-with-ResetLocale.exe

closed 2022-01-18 by marcelstoer

canedje, 2022-02-20:
This latest version "resetLocal" is working again. Thanks

user897943, 2022-03-20:
Not working for me. Windows 11 22572 build. Yes I'm running the "resetlocale" version.

marcelstoer, 2023-04-18:
Thanks for heads up Artem. No idea why the patched file vanished from my server. It's fixed now.

commit eef9f3a5 (2022-01-18): Fix the Windows + Python 3.8+ + wxPython 4.1 locale issue

Fixes #78