marcelstoer/nodemcu-pyflasher #80: Bug: Flash mods info popup not clearing itself

Ashfaaq18, 2021-09-25:
As shown in the picture below, when I hover over the flash mods info icon, the info menu pops up. 
Once the cursor moves away from the info icon, the info menu does not clear itself. 
Even moving or minimizing the window does not clear the info menu.

[image]

My win10 specs:
version: 20H2
OS build: 19042.1237

marcelstoer, 2021-09-25:
On my system I can click the overlay (or even the main window) to make it disappear.

Ashfaaq18, 2021-09-25:
I'm assuming you tested under windows. If yes, then yeah, kinda strange. 
I tried restarting my pc and running the app again. Still the same issue. Not sure if a windows update messed this up.

marcelstoer, 2021-10-21:
I can confirm this to be an issue with PyFlasher v5 on a recent Windows 10 system.

chall3ng3r, 2021-11-15:
Yes, latest build have this issue on Windows 11 as well. Popup does not disappear once open.

0xenca, 2022-07-27:
Confirmed Windows 10 still 2022
[image]

JonnyPhenomenon, 2022-12-29:
HAH! yeah, that pop over is annoying. I've got three of them on my screen as I type this.  plus the three in the screen shot above! lol!

marcelstoer, 2024-07-14:
Actually appears to be an incarnation of issue https://github.com/wxWidgets/Phoenix/issues/2554.

marcelstoer, 2025-01-31:
v5.1 should fix this, please verify, thanks.

https://github.com/marcelstoer/nodemcu-pyflasher/releases/tag/v5.1.0

closed 2025-01-31 by marcelstoer