marcelstoer/nodemcu-pyflasher #92: ESP32-C3 Problem - HELP

lucasromeiro, 2022-08-20:
Hello,

I'm having trouble recording an ESP32-C3.
I'm using a NOVA development board. The firmware is not written and the screen looks like this.
Any tips on how to resolve this?

[image]

marcelstoer, 2022-08-20:
esptool 3.0 maybe doesn't support the C3 yet: https://github.com/espressif/esptool/releases/. A new PyFlasher that includes the latest esptool is in the making.

lucasromeiro, 2022-08-21:
> esptool 3.0 maybe doesn't support the C3 yet: https://github.com/espressif/esptool/releases/. A new PyFlasher that includes the latest esptool is in the making.

Thanks!!!
How long will it take to release the new version?

Schallschutz, 2023-04-14:
Hello are there any news regarding this issue ?

closed 2026-10-12 by marcelstoer