marcelstoer/nodemcu-pyflasher #93: Failing to upload new ESP8266 ESP12E Firmware using the PyFlasher tool.

kudarukuni, 2022-08-27:
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\PYFLASHER\Main.py", line 101, in run
    esptool.main(command)
  File "C:\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\__init__.py", line 637, in main
    esp = esp or get_default_connected_device(
  File "C:\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\__init__.py", line 880, in get_default_connected_device
    _esp = detect_chip(
  File "C:\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\cmds.py", line 91, in detect_chip
    detect_port.connect(connect_mode, connect_attempts, detecting=True)
  File "C:\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 603, in connect
    last_error = self._connect_attempt(
  File "C:\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 544, in _connect_attempt
    self.sync()
  File "C:\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 407, in sync
    val, _ = self.command(
  File "C:\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 337, in command
    self.write(pkt)
  File "C:\AppData\Local\Programs\Python\Python310\lib\site-packages\esptool\loader.py", line 289, in write
    self._port.write(buf)
  File "C:\AppData\Local\Programs\Python\Python310\lib\site-packages\serial\serialwin32.py", line 325, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\AppData\Local\Programs\Python\Python310\lib\threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "C:\Desktop\ESP8266 FLASH\PYFLASHER\Main.py", line 108, in run
    self._parent.report_error(e.strerror)
  File "C:\Desktop\ESP8266 FLASH\PYFLASHER\Main.py", line 383, in report_error
    self.console_ctrl.SetValue(message)
TypeError: TextEntry.SetValue(): argument 1 has unexpected type 'NoneType'

marcelstoer, 2022-08-27:
The app can't do anything about the `SerialTimeoutException` but could guard against writing a `NoneType` to a text control.

kudarukuni, 2022-08-27:
Someow i did manage to upload the firmware and I used nodemcu_float_0.9.6.bin

marcelstoer, 2022-08-27:
> I used nodemcu_float_0.9.6.bin

Don't - that's ancient, 7(?) years old: https://nodemcu.readthedocs.io/en/release/getting-started/

closed 2022-08-27 by marcelstoer