marcelstoer/nodemcu-pyflasher #96: Upload LittleFs.bin

peakpeak-github, 2023-02-14:
Nice tool! It would be nice if I also could upload the filesystem (littlefs.bin) with it.
Any plans for that?

marcelstoer, 2023-02-14:
There are no plans for that as this feature wouldn't be addressing the same user group the app was built for. IMO it already has too many knobs 😉 I assume if you know how to build a filesystem binary you are expert enough to also use a tool more geared towards exports?

Maybe if the GUI had some sort of "Advanced" section (collapsed by default) one could add a filesystem section there. I would certainly consider this if someone offered a PR to that effect.

That being said I already built something similar earlier. For my company ThingPulse we forked this project to build the [App Fairy](https://github.com/ThingPulse/app-fairy). Explanation in our blog at https://thingpulse.com/install-configure-apps-with-the-app-fairy/

It uses `spiffsgen.py` from Espressif to [build a SPIFFS image](https://github.com/ThingPulse/app-fairy/blob/master/Main.py#L119) containing the application properties entered in the GUI. That binary is then flashed after the app binary.

peakpeak-github, 2023-02-15:
Solved it with esptool.py which is perfectly capable of uploading a complete image with both program section and filesystem section.

Thank you!

closed 2025-02-18 by marcelstoer