marcelstoer/nodemcu-pyflasher #103: Build nodemcu-pyflasher in 2024

Doohan23, 2024-03-10:
Hello,

I've been trying to build nodemcu-pyflasher to run it in Windows to program an ESP8266.
I found some issues, but finally I managed to do it.

Just if it's helpful for anyone, this are the main points to do it:

- Use Python 3.10. To be more specific, I use Python 3.10.11
- Use PyInstaller 4.10. Requirements file indicates 4.8
- Use wxPython 4.2.1. Requirements file indicates 4.1.1
- Use Espytools 3.3.3. Requirements file indicates equal or greater than 3.0 but, so the version 4.x.x was installed, but it gives an error of an incorrect path or non existing file during uploading.

Hope it helps!

amelted, 2024-07-08:
I used pyinstaller 6.0 and it worked, anything lower wouldnt. 
i didnt really change anything else
side note, i did have to change one of the build params in the script but it seems inconsequential
I'm on python 3.12.4

DeeEmm, 2024-11-28:
I pushed all of the dependencies to their most current versions and everything is working fine here

my requirements.txt

```
esptool>=3.0    
pyserial~=3.5
wxPython~=4.2.2
PyInstaller~=6.11.1
httplib2>=0.22.0
pyinstaller-versionfile>=2.0.0
```

Also had to remove the **--windowed \** build parameter from build.sh as it appears that this has now been depreciated.

marcelstoer, 2025-01-31:
Available as v5.1, please verify, thanks.

https://github.com/marcelstoer/nodemcu-pyflasher/releases/tag/v5.1.0

closed 2025-01-31 by marcelstoer