marcelstoer/nodemcu-pyflasher #107: Virus?

5p0ng3b0b, 2025-02-03:
FYI the windows exe downloaded here is flagged as a virus/malware/trojan.
https://www.virustotal.com/gui/file/93ac8d5c4fd11eb098cf7fc628accf2ddfbd69ad52437a0b0e60a14246238260

marcelstoer, 2025-02-03:
See the other virus issues: https://github.com/marcelstoer/nodemcu-pyflasher/issues?q=is%3Aissue%20state%3Aclosed%20virus

From https://coderslegacy.com/pyinstaller-exe-detected-as-virus-solutions/
- we don't use `onefile`
- we build the PyInstaller bootloader manually
- we don't use code signing cert because...who's gonna fund that for a free piece of software?

closed 2025-02-03 by marcelstoer

marcelstoer, 2025-02-03:
FWIW I already sent "false positive" reports to Avast and GData. The community can help by doing the same.

5p0ng3b0b, 2025-02-05:
Thank you for your efforts. Yeah I gathered it was a false positive .

marcelstoer, 2025-02-05:
We're now down to 10/71 on VirusTotal. Apart from Google and TrendMicro all the major ones I contacted, updated their rules. Fortinet even did it within 50min(!).